MESAInternational/B2MML-BatchML #1: Handling released versions

Dennis-Brandl, 2016-05-06:
I believe that we need to change the directory structure, with a directory for each version. This way we won't have hundreds of files for multiple versions. 
Suggestion:
  B2MML-BatchML-V0600
  B2MML-BatchML-V0601-RC
  B2MML-BatchML-V0601
  B2MML-BatchML-V0700-RC

Emers, 2016-05-06:
I agree with Dennis.  This way we can create new directories as we take on new versions.  I assume then issues will be associated with specific releases, this will let us track issues on released versus under development versions.

JoeDo, 2016-05-06:
I agree that we should consider releases. However, the approach I would recommend would be to use the **Release** function of GitHub, which _tags_ a certain commit with a release number. In this case, I would tag the initial commit as something like "V0600". From that point on, the Master branch would always contain the latest updates. When we're prepared to issue another release, we just create a new release (e.g., V0601-RC).

If we're in agreement, I'd be happy to set this up.

Dennis-Brandl, 2016-05-06:
The problem is that we have started a convention of having the version in the file name, and in the schemas themselves.  This was the recommended practice to ensure that different version files would not get mixed up.  The next version would have had V0601 appended to the name of the file.  If we go with the Release function, then we would have to change the names in the files to remove the version number, right?

JoeDo, 2016-05-12:
(@Dennis-Brandl, Not sure if you meant to close this issue, but I think it's still worth discussing, so I've reopened it.)

Correct -- If we were to go this route, we would want to change the filenames and references within the schema. If we're not comfortable doing that at this time, we can just change the directory structure for now.

Dennis-Brandl, 2016-05-12:
Sorry I didn't mean to close it. I hit the wrong button :-( 
I'm not comfortable changing the file naming convention, but what we could do is use the **Release** function for the release candidates.  It was always a pain to have to rename all of the files when we changed a release candidate. We should use the directory route to create a V0600, and a V0601 (or V0700 depending on what Dave wants), then do all of the release candidates under V0601, until we generate the official V0601 release.  
Will this method work?

JoeDo, 2016-05-12:
Haha, no worries. And yes, it should work. It's uncommon to have version numbers in file names for source code, but I understand we have historical reasons for doing this.

Feel free to introduce the directory structure as you described.

closed 2016-05-12 by JoeDo