MESAInternational/B2MML-BatchML #4: Example BatchML

vaibhavkhanduja, 2016-07-17:
Hi,

Is there a place where I can find example BatchMLs? I am writing code and want to test my parser.

Thanks,

vaibhavkhanduja, 2016-07-20:
@BrentHumphreys @JoeDo @Emers @agileteams @jvieille Any thoughts here?

Emers, 2016-07-20:
Hi,
Unfortunately we only have a couple of examples, and these are for older versions.  We are also using a Yahoo Group as a general list server (we are still in progress of fully moving onto GitHub), I've posted a request to that list server.  You can use this URL to see any responses - I administer the Yahoo Group so if you make a request to join I'll accept it.
https://groups.yahoo.com/neo/groups/mesa-xml-wg/conversations/messages

Dave Emerson - emers

vaibhavkhanduja, 2016-07-20:
@Emers Thanks I have sent you a request to join the group.

closed 2019-02-11 by Dennis-Brandl