MESAInternational/B2MML-BatchML #6: RangeType value type

jrcroteau, 2016-10-13:
Section 4.8 of ANSI/ISA-95.00.02-2010 (IEC 62264-2 Mod) suggests a value type called RangeType but I do not see specific support for this in B2MML. Is this something that is a gap or accounted for in another way?

Dennis-Brandl, 2016-10-13:
Section 4.8.2 defines how complex value types can be encoded, if they are represented as strings. Basically there were many B2MMl and ISA 95 comments on how values, which were more than single values (such as arrays and structures), were to be represented.  Section 4.8.2 is an example of encoding complex data into strings (JSON like) and are the rules for encoding these within the ValueString element in the ValueType element.  Maybe not the best XML representation, but we never extended the ValueType to handle these complex situations.  Probably because they constituted a tiny subset of exchanged data, but would make the schemas really really complex.

jrcroteau, 2016-10-16:
Is that where we plan to leave it then? I saw in the comments section of the BHP schemas used as a basis for B7 that they added a "RangeType". Will that be left out of the official schema?

Dennis-Brandl, 2016-10-17:
We usually wait until we hear something official from the ISA 95 committee. Since the BHP work is still a working draft, not even a committee draft yet, I think we just have to see what comes out of the committee.  I want to point out that BHP is not a MESA member, so their work is currently their own private version.  We can merge their changes in later, but we don't want to be too far ahead of the ISA 95 committee decisions.

closed 2020-01-27 by Dennis-Brandl