MESAInternational/B2MML-BatchML #8: Typo in Material XSD

Dennis-Brandl, 2016-10-18:
-----Original Message-----
From: Bernhard Wally [mailto:wally@big.tuwien.ac.at]
Sent: Wednesday, September 14, 2016 2:26 AM
To: b2mml@mesa.org
Subject: Erratum B2MML V0600

Dear Sir or Madam,

I'd like to point you to a typo in the Material Information Schema (XSD) of the downloadable B2MML Specification.
File: B2MML-V0600-Material.xsd
Line: 246

<xsd:element name = "AssemblylDefinitionID"
-----------------------------^

should probably say (without the "l" after "Assembly"):
<xsd:element name = "AssemblyDefinitionID"

This typo also propagates to the corresponding documentation:

File: B2MML-V0600-Material.pdf
Page: 7 (Page 9 in PDF Viewer)

The figure showing internals of MaterialDefinition also says "AssemblylDefinitionID"
instead of
"AssemblyDefinitionID".

This error/typo was introduced in V0500.

Also, look for the String "B2B2MML" on the corresponding Webpage <https://services.mesa.org/ResourceLibrary/ShowResource/0f47758b-60f0-40c6-a71b-fa7b2363fb3a>

Should all be easy to fix =)

Best Regards, Bernhard Wally

--
Bernhard Wally

Business Informatics Group
Institute of Software Technology and Interactive Systems

Vienna University of Technology
Favoritenstrasse 9-11/188
1040 Vienna, Austria

T: +43 1 58801 188695
F: +43 1 58801 9 188695

http://www.big.tuwien.ac.at

Visit us at STAF'16, July 4-8 2016
http://staf2016.conf.tuwien.ac.at/

closed 2016-10-28 by JoeDo

Dennis-Brandl, 2016-10-28:
Thanks Joe. I was trying to follow the proper method.

JoeDo, 2016-10-28:
Yep, this was perfect. I merged the Pull Request into Master. Feel free to delete the Dennis-Brandl-patch-1 branch if you don't need it.