MESAInternational/B2MML-BatchML #11: Assign LICENSE

jpdillingham, 2017-10-06:
The following license text is provided within ```AllSchemas.xsd```:

```
        This MESA International work (including specifications, documents, 
        software, and related items) referred to as the Business To 
        Manufacturing Markup Language (B2MML) is provided by the copyright 
        holders under the following license. 

        Permission to use, copy, modify, or redistribute this Work and its 
        documentation, with or without modification, for any purpose and 
        without fee or royalty is hereby granted provided MESA International 
        is acknowledged as the originator of this Work using the 
        following statement:

        "The Business To Manufacturing Markup Language (B2MML) is used 
        courtesy of MESA International."

        In no event shall MESA International, its members, or any 
        third party be liable for any costs, expenses, losses, damages or 
        injuries incurred by use of the Work or as a result of this 
        agreement.
```
This copy should be inserted into a LICENSE (no extension) file in the root of the repository.

jpdillingham, 2017-10-06:
I'll issue a PR shortly.

JoeDo, 2018-05-22:
Merged -- thanks, @jpdillingham!

closed 2018-05-22 by JoeDo