MESAInternational/B2MML-BatchML #13: B2MML Library Development

jpdillingham, 2018-01-16:
Continuing a discussion from [here](https://github.com/jpdillingham/B2MML.NET/pull/3).  @BrentHumphreys, @feldrim

I'd like to kick off this thread by suggesting that we first understand the use case(s) for such a library.  I initially created it purely as a target for serialization and I think this is still the primary case, however we should get into alignment.

feldrim, 2018-01-20:
1. The main purpose of B2MML library is to serialize the data into B2MML and deserialize it to the objects. But when the developers have these as classes, it's a possible use case that they use this as a shared library of domain entities in their project. So that B2MML schemas could become the data model namespace / package easily. So if any software development team has used B2MML in this way, it should be learnt how did they implemented it and what obstacles they met. Probably it's done by beginning with B2MML auto-generated classes but modified after for business requirements.

The question is since B2MML is designed as an "interface" by means of software, should B2MML library use as data models be considered as a bad practice? Or should that be taken into account?

2. Serialization to JSON in web service world is a must in 2018, while I am even considering [Protobuf](https://github.com/mgravell/protobuf-net)  and [ZeroFormatter](https://github.com/neuecc/ZeroFormatter) for less overhead on network.

This made me think about all the attributes on the classes and properties generated, since the classes muat be easily serialized to JSON, proto or ZeroFormatter objects. How would that effect the end result? Should that be counted within the serialization purpose? If so, what should be the correct way?

I guess that's all for now. I have questions but not answers for any of them. Thank you for opening the issue @jpdillingham , it's a great opportunity to discuss this matter.

Dennis-Brandl, 2018-01-22:
B2MML was not designed to serialize local data into B2MML format.  Instead it was designed to provide a common format, that local is converted into, and converted back from. 

That being said, there are a lot of people using the ISA 95 and ISA 88 exchange models as the starting point for their own local definitions.  The models in the standard need a lot of extra attributes, objects, and relationships to have a complete local definition.  

So in regards to your question “should B2MML library use as data models be considered as a bad practice?”, I think it is not a bad practice and I think we need to provide some documentation on how they can convert B2MML into their initial object definitions. 

 

Regarding the attributes and JSON. Most of the attributes are used to support the UN/CEFACT information, which I believe is little real value. The only attribute we regularly use is the OtherValue used on fixed value enumerations, to support extensibility.  If anyone can come up with an alternative using JSON, we should use it. 

 

Dennis Brandl

208 Townsend Ct, Suite 200

Cary, NC 27518-8319, USA

+1-919-852-5322 (Office)

+1-919-656-2205 (Cell)

+1-832-201-0554 (Fax)

 <http://www.brlconsulting.com/> www.brlconsulting.com   

    <mailto:DnBrandl@BRLConsulting.com> DnBrandl@BRLConsulting.com

JoePlant, 2018-01-22:
This is an interesting discussion point.  I personally have used the B2MML schemas as the messaging format based rather than an interface format for services to be implemented.

My experience has been both from a software developer and also for implementation in projects .  In both instances, I have not wanted to implement the complete B2MML messaging structure.
I find that the definition helps me to understanding the types of data and the format of how they could be implemented.   When some information needs to be exchanged, it can help provide the structure although there is usually some complexity in the implementation given trying to use the attributes in any generated code.

The message schema (the data that needs to be exchanged) and the interface schema (methods to be called) are quite verbose and seem to expect a full interface to be implemented.  This is probably a challenge to people who want to implement some or all of it as there is a lot of complexity required to implement.

feldrim, 2018-02-04:
I'd like to thank everyone on this topic for the precious discussion, especially to @Dennis-Brandl for replying my naive questions in an elightening and brief manner as a senior.

So, I believe, we can discuss about the possible use cases for the B2MML libraries. In what ways the library would help developers? I used them as my entities (or models, whichever you prefer) in project and interfaces among services. What else is possible and under what conditions it's recommended to use or not to use the library in that use case?

Thanks to @JoePlant , I've found the courage to ask some other question. Yet it's verbose in the B2MML.NET library we can find 1200+ classes ready to use but isn't this too much? Can't we seperate them in namespaces/packages etc. (whatever the language calls the unit) based on the XSD in a way that developers can use the whole package or partially?

BrentHumphreys, 2018-02-05:
I would like to second the suggestion to split out the namespaces.   Right now, all of B2MML is in a single namespace, even though their are separate files that do a very good job of identifying a way to break them up.  I had forked JP's repo and toyed with the same idea in .net, but it would make much more sense to do it in the schema.
https://github.com/BrentHumphreys/B2MML.NET/tree/master/B2MML.NET

feldrim, 2018-05-18:
It's been a while and I'm back before using B2MML.Net in a new project.  I'm still stuck in the same philosophical question: What is the purpose of the B2MML libraries?

Because all other questions are children of that and the project roadmap shall be designed accordingly. Via the classes, what should the developers achieve? Base classes for business entities? Or any other implementation? What is the suggested use?

I believe a tutorial written in any language based on this purpose will be more useful than a simple readme. When a new library published, the tutorial must take part in the documentation, at least in one language just to explain. Most probably a simple "Hello World"-ish part of code in a separate repository would help other developers add other versions also (For example when a Java sample added, I would be voluneer to add C# version if it).

Emers, 2018-05-23:
Hello MESA XML Committee,

Since the March committee meeting some v7 changes added to our GitHub repository but our volunteers have been busy with day jobs and much more remains to be done.

I will set up a web meeting for July 5, 2018 08:00 am (US Central time).   Prior to that meeting I urge everyone to make a fork and add your edits to GitHub.  When you have completed your edits, or are at a logical break, make a pull request.  The July 5 web meeting will be a working meeting to review the pull requests, decide whether to include each edit and create a new master branch.

We will repeat the fork/pull request/review on a roughly monthly cycle until the committee decides the master branch is complete enough to be a release candidate for v7.

Everyone that wants to contribute to B2MML v7 or speed up its release should be on GitHub making edits to a fork.   Feel free to use the XML Committee list server (xml-sub-committee@mesa.org<mailto:xml-sub-committee@mesa.org>) or GitHub's Issue's feature to let others know what you will be editing or to ask questions about edits.

If you have questions about using the B2MML GitHub repository Joe Dolivo our repository administrator is willing to help you.

Best Regards,

Dave Emerson
MESA XML Committee Chair
+1.469.774.1239 - mobile
e-mail: dave.emerson@us.yokogawa.com<mailto:dave.emerson@us.yokogawa.com>

CONFIDENTIAL: This message (including any attachments) is confidential and may be privileged and is intended only for the review of the party to whom it is addressed. If you have received it by mistake, please do not read it, notify the sender by return e-mail and delete this message from your system. Any unauthorized use, dissemination, disclosure or copying of this communication in whole or in part thereof is strictly prohibited and may be unlawful.
Please note that e-mails are susceptible to change. YOKOGAWA shall not be liable for the improper transmission of this communication, any damage to your system nor reliance upon it by any other person or for any other purpose. YOKOGAWA does not guarantee that this communication is free of viruses, interceptions or interference. YOKOGAWA thanks you for your cooperation.

Dennis-Brandl, 2020-01-27:
No actions have been taken, or further discussion on this. 
I will close the thread/issue. If it comes up again we can raise a new issue.

closed 2020-01-27 by Dennis-Brandl