MESAInternational/B2MML-BatchML #17: Removal of Production schemas

andreb1972, 2019-02-04:
https://github.com/MESAInternational/B2MML-BatchML/commit/f298399f1b18d1da2fe494ef046f84f201bb14e3#commitcomment-32176946

The LocationType element in Common.xsd is only used in Production schemas so can be removed.

In Extensions.xsd there are references to the Production schemas that can be removed. Sample:
<xsd:group name="ProductDefinition">
xsd:sequence

</xsd:sequence>
</xsd:group>

Dennis-Brandl, 2019-02-11:
Based on the February 11, 2018 meeting, this will be fixed. In the future we will ask the submitter to make a PULL request with the patch.

Dennis-Brandl, 2020-01-27:
To be fixed in the January 2020 Sprint

JoeDo, 2020-02-03:
Confirmed that LocationType and ProductDefinition have been removed.

@Dennis-Brandl, there are other Product-related schema in B2MML-Extensions.xsd, such as ProductInformation and ProductSegment. Also, line 384 still references the ProductDefinition schema. Do all of these references need to be removed as well?

Dennis-Brandl, 2020-08-13:
Fixed in the soon to be released version, issue closed.

closed 2020-08-13 by Dennis-Brandl