MESAInternational/B2MML-BatchML #18: CommonExtensions .xsd

andreb1972, 2019-02-04:
Typo in CommonExtensions.xsd and also in OperationsEvent.xsd

The group name in CommonExtensions should be EventClassProperty
 <xsd:group name = "OperationsEvenClassProperty">
                <xsd:sequence>
                        
                </xsd:sequence>
        </xsd:group>

The reference to this type is in OperationsEvent.xsd:
<xsd:complexType name="OperationsEventClassPropertyType">
	<xsd:group ref="Extended:OperationsEvenClassProperty" minOccurs="0" maxOccurs="1"/>

Dennis-Brandl, 2019-02-11:
Based on the February 11, 2018 meeting, this will be fixed. In the future we will ask the submitter to make a PULL request with the patch.

Dennis-Brandl, 2020-01-27:
To be fixed in the January 2020 Sprint

JoeDo, 2020-02-03:
Confirmed that the change has been made in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-13:
Corrected, closing issue.

closed 2020-08-13 by Dennis-Brandl