MESAInternational/B2MML-BatchML #20: Common.xsd - LocationType

andreb1972, 2019-02-04:
Consider removing LocationType from Common.xsd.  It was previously used in the production schemas which have been removed

<xsd:complexType name="LocationType1">
		<xsd:sequence>
			<xsd:element name="EquipmentID" type="EquipmentIDType"/>
			<xsd:element name="EquipmentElementLevel" type="EquipmentElementLevelType"/>
			<xsd:element name="Location" type="LocationType1" minOccurs="0"/>
			<xsd:group ref="Extended:Location" minOccurs="0" maxOccurs="1"/>
		</xsd:sequence>
</xsd:complexType>

Dennis-Brandl, 2019-02-11:
Based on the February 11, 2018 meeting, this will be fixed. In the future we will ask the submitter to make a PULL request with the patch.

Dennis-Brandl, 2020-01-27:
To be remove din the January 2020 Sprint

JoeDo, 2020-02-03:
Confirmed that this type does not exist in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-13:
Closing the issue.

closed 2020-08-13 by Dennis-Brandl