MESAInternational/B2MML-BatchML #23: MasterDataProfile.xsd - Typo IDandDescriptionType

andreb1972, 2019-02-04:
The element name attribute has a typo:
`	<xsd:complexType name="IDandDesciptionType">`

IDandDesciptionType = IDandDescriptionType

Elements that reference this type will also need to be corrected:

```
<xsd:complexType name="MasterDataProfileType">
		<xsd:sequence>
			<xsd:element name="Header" type="MasterDataProfileHeaderType"/>
			<xsd:element name="UnitOfMeasure" type="IDandDesciptionType" minOccurs="0" maxOccurs="unbounded"/>
			<xsd:element name="DataType" type="IDandDesciptionType" minOccurs="0" maxOccurs="unbounded"/>
			<xsd:element name="ValueKey" type="IDandDesciptionType" minOccurs="0" maxOccurs="unbounded"/>
```

Dennis-Brandl, 2019-02-05:
I failed spelling, yes it should be IDandDescriptionType.

Dennis-Brandl, 2020-01-27:
To be fixed in the January 2020 Sprint

JoeDo, 2020-02-03:
Confirmed that the change has been made in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-13:
Closing issue.

closed 2020-08-13 by Dennis-Brandl