MESAInternational/B2MML-BatchML #24: TestSpecification.xsd - Optional ID

andreb1972, 2019-02-04:
In all (or most) other schemas, the ID of the primary entity is always a required element.  In the TestSpecification,  this ID is optional.
```
<xsd:complexType name="TestSpecificationType">
		<xsd:sequence>
			<xsd:element name="ID" type="IdentifierType" minOccurs="0"/>
```

TestSpecificationID is referenced in other schemas (like ProcessSegment and OpsPerformance), so consider making the ID a required element
```
<xsd:complexType name="TestSpecificationType">
		<xsd:sequence>
			<xsd:element name="ID" type="IdentifierType" minOccurs="1"/>
```

Dennis-Brandl, 2019-02-05:
Yes, looks like a bug/type. I believe it should be required. (actually just remove the "minOccurs")

Dennis-Brandl, 2019-02-11:
Based on the February 11, 2018 meeting, this will be fixed. In the future we will ask the submitter to make a PULL request with the patch.

Dennis-Brandl, 2020-01-27:
To be fixed in the January 2020 Sprint

JoeDo, 2020-02-03:
Confirmed that the change has been made in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-13:
Closing issue.

closed 2020-08-13 by Dennis-Brandl