MESAInternational/B2MML-BatchML #25: OperationalLocation.xsd - OptionalID

andreb1972, 2019-02-04:
In OperationalLocation.xsd, the ID of OperationalLocationType is optional:
```
<xsd:complexType name="OperationalLocationType">
		<xsd:sequence>
			<xsd:element name="ID" type="IdentifierType" minOccurs="0"/>
```
In the OperationalLocationClassType element, the OperationalLocationID is used to reference the location.  
```
<xsd:complexType name="OperationalLocationClassType">
		<xsd:sequence>
			<xsd:element name="IncludesOperationalLocationClassID" type="IdentifierType" minOccurs="0" maxOccurs="unbounded"/>
			<xsd:element name="OperationalLocationID" type="IdentifierType" minOccurs="0" maxOccurs="unbounded"/>
```
Consider making the element required, as it is used as a record identifier.
```
<xsd:complexType name="OperationalLocationType">
		<xsd:sequence>
			<xsd:element name="ID" type="IdentifierType" minOccurs="1"/>
```

Dennis-Brandl, 2019-02-05:
Yes, looks like a bug/type. I agree it should be required.

Dennis-Brandl, 2019-02-11:
Based on the February 11, 2018 meeting, this will be fixed.  In the future we will ask the submitter to make a PULL request with the patch.

Dennis-Brandl, 2020-01-27:
To be fixed in the January 2020 Sprint

JoeDo, 2020-02-03:
The definition of OperationLocationType has been moved from OperationLocation.xsd to Common.xsd in the January-2020-Sprint branch, and the "minOccurs" attribute has been removed. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-13:
Fixed in soon to be released version. Issue closed.

closed 2020-08-13 by Dennis-Brandl