MESAInternational/B2MML-BatchML #30: StorageLocation tag definition inconsistent inside MaterialLot, MaterialRequirement

racheliurui, 2019-03-19:
I am not quite sure about the difference between "ResourceLocationType" and "SotrageHierarchyScopeType" but it's inconsistent while being referred as the "MaterialLot/StorageLocation" and "MaterialRequirement/StorageLocation".

We have used the ResouceLocationType with MaterialRequirement, but then the "MaterialLot/StorageLocation" only support a single string ID.   Consider the tag name is the same, it's causing confusion while doing the interface and data aggregation design.

andreb1972, 2019-05-14:
I had a look at this, and you are correct - the naming of the existing ResourceLocationTypes does conflict with the new OperationalLocation.  In each of the requirement elements we have:

PersonnelRequirement = OperationalLocation
EquipmentRequirement=OperationalLocation
PhysicalAssetRequirement=PhysicalLocation
MaterialRequirement =StorageLocation

Each of these Requirement elements should now also have an element representing the OperationalLocationType. 

As the OperationalLocation can exist on its own i.e. it has transactions etc. it should be referenced by ID, rather than a nested object... something like EquipmentID:
```xml
<xsd:complexType name="OpEquipmentRequirementType">
		<xsd:sequence>
                        ...
			<xsd:element name="EquipmentID" type="EquipmentIDType" minOccurs="0" maxOccurs="unbounded"/>
			<xsd:element name="OperationalLocationID" type="OperationalLocationIDType" minOccurs="0" maxOccurs="unbounded"/>
                        ... 
		</xsd:sequence>
	</xsd:complexType>
```
If the OperationalLocation is somewhat dynamic, as in the resources sector, then could be an issue, as the operational location is defined at the time of scheduling/actualizing.  In this case a the OperationalLocation and the SegmentRequirement would need to be included in the same payload, so perhaps an OperationsEvent message construct would be required.

Alternatively, the OperationalLocation could be included as a nested type, which would allow a reference by ID or the entire OperationalLocation could be described within the Requirement element.   The issue then would be naming, so as to maintain backward compatibility"

```xml
<xsd:complexType name="OpEquipmentRequirementType">
		<xsd:sequence>
                        ...
		          <xsd:element name="OperationalLocation" type="OperationalLocationType" minOccurs="0" maxOccurs="unbounded"/>
                        contradicts
                           <xsd:element name="OperationalLocation" type="ResourceLocationType" minOccurs="0"/>
                        ... 

		</xsd:sequence>
	</xsd:complexType>
```
This approach would require some restructure of the XSD's, to make the OperationallocationType a common type

vitalJP, 2019-10-10:
Adding additional information:

### Background
>complexType name = "MaterialSubLotType", element name ="StorageLocation"; Change type = "StorageHierarchyScopeType" to "ResourceLocationType" to align with updated ISA-95 Part 2 Material SubLot Object.

### Impacted Types

B2MML - Material.xsd
- MaterialLotType
- MaterialSubLotType

### Supporting Document

ISA-950002
Section 5.7.6 and 5.7.8

Dennis-Brandl, 2020-01-27:
The proposal is to: 
1) Move the OperationalLocationType to the B2MML_Common file. 
2) Remove StorageHierarchyScopeType and replace with ResourceLocationType
3) Add a <choice> in ResourceLocationType to allow selection of either:
    a)  A Location ID, which is a reference to an OperationalLocation that is previously defined.
    b)  A Location of type OperationaLocationType, so that an entire OperationalLocation can be included in the ResourceLocationType
    c)  Add the following two codetypes to ResourceLocationType1Type: OperationalLocation ID and Person (to allow a resource to be associated with a person, such as a tool's location is with a specific person). 

    <xsd:complexType name="ResourceLocationType1Type">
        <xsd:simpleContent>
            <xsd:restriction base="CodeType">
                <xsd:enumeration value="Operational Location ID"/>
                <xsd:enumeration value="Operational Location"/>
                <xsd:enumeration value="Person"/>
                <xsd:enumeration value="Equipment"/>
                <xsd:enumeration value="Physical Asset"/>
                <xsd:enumeration value="Description"/>
                <xsd:enumeration value="Other"/>
            </xsd:restriction>
        </xsd:simpleContent>
    </xsd:complexType>
        
    <xsd:complexType name="ResourceLocationTypeType">
		<xsd:annotation> 
			<xsd:documentation>
Indicates whether the storage location attribute refers to an operational location, equipment 
or physical asset object, or contains a description of the storage location.
Mandatory where a storage location is specified.
Defined values are
-	Operational Location ID: storage location attribute references an operational location (OperationalLocationID);
-	Operational Location:    storage location attribute defines an operational location (OperationalLocationType);
-	Person: 				 location attribute references a Person (PersonID) (eg: material or equipment is with a person);
-	Equipment: 				 storage location attribute references an equipment object (EquipmentID);
-	Physical Asset: 		 storage location attribute references a physical asset (PhysicalAssetID); and
-	Description: 			 storage location attribute contains a description of the storage location, such as a street address.
			</xsd:documentation>
		</xsd:annotation> 
          <xsd:simpleContent>
            <xsd:extension base="ResourceLocationType1Type">
                <xsd:attribute name="OtherValue" type="xsd:string"/>
            </xsd:extension>
        </xsd:simpleContent>
    </xsd:complexType>
        
    <xsd:complexType name="ResourceLocationType">
        <xsd:sequence>
		<xsd:choice>
			<xsd:element name = "LocationID" type = "TextType" />
			<xsd:element name = "Location"    type = "OperationalLocationType" />
		</xsd:choice>
		<xsd:element name = "LocationType" type = "ResourceLocationTypeType" />
        </xsd:sequence>
    </xsd:complexType>

JoeDo, 2020-02-03:
Confirmed that (1), (2), and (3) referenced above have been made in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
To be added in the soon to be released version. Closed issue.

closed 2020-08-14 by Dennis-Brandl