MESAInternational/B2MML-BatchML #31: PhysicalAsset Requirement

andreb1972, 2019-05-14:
The OpPhysicalAssetRequirementType and OpEquipmentRequirementType have an EquipmentLevel element which is of type HeirarchyScope.   This appears to be incorrect, as there is already a HeirarchyScope within these types.  Element level is usually contained within a HS, rather than being one. Is this intended?

```xml
<xsd:complexType name="OpEquipmentRequirementType">
         <xsd:sequence>  
                ...  
                <xsd:element name="EquipmentLevel" type="HierarchyScopeType" minOccurs="0"/>  
               ...  
         </xsd:sequence>  
</xsd:complexType>  
```

Dennis-Brandl, 2019-05-20:
This is intended. The hierarchy scope identifies where the exchanged information fits within the role-based equipment hierarchy. It defines the scope of the exchanged information, such as a site or area for which the information is relevant. The hierarchy scope identifies the associated instance in the role-based equipment hierarchy.
The EquipmentLevel in the OpPhysicalAssetRequirementType and OpEquipmentRequirementType define what the actual equipment is, in the hierarchy. so the first use is for the scope of the exchanged data (such as the exchanged information is scoped to a site), the second is what the asset actually is.  we used the same type for both. Maybe a bad idea.

Dennis-Brandl, 2020-01-28:
I recommend not changing these uses, because they would be breaking changes.  I will try and clarify in the documentation on the January 2020 Sprint.

JoeDo, 2020-02-03:
Verified that the following documentation exists for HierarchyScopeType, defined in Common.xsd, in the January-2020-Sprint branch: _The hierarchy scope identifies where the exchanged information fits within the role-based equipment hierarchy. It defines the scope of the exchanged information, such as a site or area for which the information is relevant. The hierarchy scope identifies the associated instance in the role-based equipment hierarchy._

I'll circle back and Close this issue after the changes have been merged into Master.

PierreREN, 2020-08-13:
Still, with the clarification, using HierarchyScopeType for attribute "equipmentLevel" remains a very bad idea. "equipmentLevel" represents the hierarchy level where the equipment refers to and its value belongs to the enumeration { ENTERPRISE, SITE ,AREA, WORK UNIT, ...}. Thus the only attribute we need from HierarchyScopeType is its EquipmentElementLevel attribute, any other attribute may only cause confuses and troubles as HierarchyScopeType is supposed to be seen as an ENTITY. An entity possesses id and relates to a table in the database for persistence, which definitely increases workload for usage.
Though changing these uses would be breaking changes, it's always worse to disappoint users with known mistakes.
Looking forward to future moves.

Dennis-Brandl, 2023-02-09:
This was a typo, changed the EquipmentLevel to the EquipmentLevelType and moved it up to after the Equipment element.

closed 2023-02-09 by Dennis-Brandl

gerardmoloneyetpartners, 2023-03-08:
for @andreb1972 and @andylee513 benefit, the typo was documented in issue #74 and fixed recently.