MESAInternational/B2MML-BatchML #35: add DispositionType definition

racheliurui, 2019-09-17:
### Proposal

Add definition for DispositionType 

```xml
     <xsd:complexType name="Disposition1Type"> 
         <xsd:simpleContent> 
             <xsd:restriction base="CodeType"> 
                 <xsd:enumeration value="Planned"/> 
                 <xsd:enumeration value="In-Process"/> 
                 <xsd:enumeration value="Restricted"/> 
                 <xsd:enumeration value="UnRestricted"/> 
                 <xsd:enumeration value="Closed"/> 
                 <xsd:enumeration value="Other"/> 
             </xsd:restriction> 
         </xsd:simpleContent> 
     </xsd:complexType> 
     <!--     -->     
     <xsd:complexType name="DispositionType"> 
 		<xsd:annotation>  
 			<xsd:documentation> 
 Defines Planning and logistics disposition of a material lot or assembly of material lots.  
 Defined values for the disposition of a material lot are 
 - Planned: a material lot that does not yet physically exist, is assigned to an operations request (segment requirement) or work request (Part 4 object) or job order (Part 4 object);  
 - In process: the material lot is in the process of being worked on;  
 - Restricted: a material lot is not permitted for normal use due to a restriction condition.  
 	EXAMPLE 5 A material lot can be awaiting a quality decision or a material lot can be physically inaccessible. 
 - Unrestricted:  material lot is permitted for normal use without restriction; and 
 - Closed:   material lot has been reconciled as completely consumed, sold or disposed of. 
 			</xsd:documentation> 
 		</xsd:annotation>  
           <xsd:simpleContent> 
             <xsd:extension base="DispositionType"> 
                 <xsd:attribute name="OtherValue" type="xsd:string"/> 
             </xsd:extension> 
         </xsd:simpleContent> 
     </xsd:complexType> 
```

### Supporting Document
>ISA-950002
Section 5.7.6
Table 69 – Material lot attributes
Table 73 – Material sublot attributes

Dennis-Brandl, 2020-01-28:
To be fixed in the January 2020 Sprint

JoeDo, 2020-02-03:
Confirmed that the change has been made in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
To be included in the soon to be released version. Issue closed.

closed 2020-08-14 by Dennis-Brandl