MESAInternational/B2MML-BatchML #36: add Disposition attribute to MaterialLotType and MaterialSubLotType

racheliurui, 2019-09-17:
### Background

ADD element name = "Disposition", type = "DispositionType" (#35 ) to to MaterialLotType and MaterialSubLotType align with updated ISA-95 Part 2 Material Lot and Material SubLot Object. Add DispositionType to Common Schema.

### Impacted Types
B2MML - Material.xsd

- MaterialLotType
- MaterialSubLotType

### Supporting Document
>ISA-950002
Section 5.7.6 and 5.7.8
Table 69 – Material lot attributes
Table 73 – Material sublot attributes

Dennis-Brandl, 2020-01-28:
To be fixed in the January 2020 Sprint

JoeDo, 2020-02-03:
Confirmed that the change has been made in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
To be included in the soon to be released version. Issue closed.

closed 2020-08-14 by Dennis-Brandl