MESAInternational/B2MML-BatchML #37: update EquipmentType to align with ISA-950002

racheliurui, 2019-09-17:
### Proposed definition

```xml
<xsd:complexType name = "EquipmentType">
    <xsd:sequence>
      ...
      <xsd:element name = "EquipmentLevel"      type = "EquipmentLevelType"   
                                                minOccurs = "0" />
      ...
      <xsd:element name = "PhysicalAssetID"     type = "PhysicalAssetIDType"   
                                                minOccurs = "0" />
      <xsd:element name = "OperationalLocation" type = "ResourceLocationType"     
                                                minOccurs = "0" />
    ...   
    </xsd:sequence>
  </xsd:complexType>
```

### Background 
> Change element "EquipmentLevel" to be with type = "EquipmentLevelType".
> ADD element name ="PhysicalAssetID" type = "PhysicalAssetIDType" 
> ADD element name = "OperationalLocationType" type = "ResourceLocationTypeType"

### Supporting Document
>ISA-950002
Table 43 – Equipment relationship roles
Table 44 – Equipment attributes

### Impacted Type

B2MML-Equipment.xsd
- EquipmentType

Dennis-Brandl, 2020-01-28:
Recommendations:
1) "Change element "EquipmentLevel" to be with type = "EquipmentLevelType".  I respectful disagree there no real reason givem. The HierarchyScopeType is already the nested type using the equipment levels. The only change would be the name of the type. Which is not relevant for generated documents.
2) "ADD element name ="PhysicalAssetID" type = "PhysicalAssetIDType".  Yes, but the type is IdentifierType to line up with the simplifcation of types and removal of intermediate types. 
3) "ADD element name = "OperationalLocationType" type = "ResourceLocationTypeType". Yes to be fixed in the January 2020 Sprint.

JoeDo, 2020-02-03:
Confirmed (1) and (2). @Dennis-Brandl, regarding (3), EquipmentType as defined in Equipment.xsd  contains an OperationalLocation element defined as type ResourceLocationType, as requested in the original proposal. Can you confirm that this is intentional and correct?

Dennis-Brandl, 2020-08-14:
In the soon to be released version the following changes were made to B2MML-Equipment.xsd:
- Added element name ="PhysicalAssetID" type = "PhysicalAssetIDType". 
- Added element name = "OperationalLocationType" type = "ResourceLocationTypeType".
- Changed EquipmentLevel type to be EquipmentLevelType. 

Issue closed.

closed 2020-08-14 by Dennis-Brandl