MESAInternational/B2MML-BatchML #38: add MaterialDefinitionPropertyID to MaterialLotPropertyType to align with ISA-950002

racheliurui, 2019-09-17:
### Background
There is 0..* map relationship from MaterialLotProperty  to MaterialDefinitionProperty. The mapping relationship is missing from the current B2MML schema. 

```xml 
<xsd:complexType name = "MaterialLotPropertyType">
...
   <xsd:element name = "MaterialDefinitionPropertyID" type = "IdentifierType"  minOccurs = "0" maxOccurs = "unbounded"/>
...
</xsd:complexType>
```

### Supporting Document
>ISA-950002
>Table 70 – Material lot property relationship roles

### Impacted types

B2MML-Material.xsd
- MaterialLotPropertyType

Dennis-Brandl, 2020-01-28:
Yes, but it should be maxOccurs = "1", as per table 70.
To be fixed in the January 2020 Sprint.

JoeDo, 2020-02-03:
Confirmed that the change has been made in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
Fixed in the soon to be released version. Issue closed.

closed 2020-08-14 by Dennis-Brandl