MESAInternational/B2MML-BatchML #39: add Required ID to each Op(Resource)CapabilityType in Common Schema

racheliurui, 2019-09-17:
### Proposed change

add element ID to impacted definitions.

```xml
<xsd:element name="ID" 				type="IdentifierType"/>
```

### Background
The current B2MML Common Schema, makes updating or deleting an EquipmentCapability extremely hard to implement.
During implementation, we found that the EquipmentCapability does not have an ID defined.
But when we retrieve the equipment capability from a time accounting system, we do see an ID is attached to each of the reported equipment capabilities. But we can't map that id to the model.
Then when the source system reports an equipment capability change, for example, a previously reported equipment capability item now needs to be deleted. How we achieve that?

After reviewing the original ISA-95 document,  we found this missing ID is not aligned with updated 950002, each resource capability object in Clause 6.4 Operations capability information has a required ID attribute.

### Current workaround

The current workaround is, we put the id in WorkCapability/ID, then every WorkCapability only contains one single EquipmentCapability.

### Supporting Document
>ISA-950002
Table 228 – Personnel capability attributes
Table 232 – Equipment capability attributes
Table 236 – Physical asset capability attributes
Table 240 – Material capability attributes

### Impacted Definitions
B2MML-Common.xsd

- OpEquipmentCapabilityType
- OpPersonnelCapabilityType
- OpPhysicalAssetCapabilityType
- OpMaterialCapabilityType

Dennis-Brandl, 2020-01-28:
This is a breaking change (since the ID is required), but it will be fixed in the January 2020 Sprint.

JoeDo, 2020-02-03:
Confirmed for the following types in the January-2020-Sprint branch:
- OpPhysicalAssetCapabilityType

@Dennis-Brandl, there appear to be similarly named types that are missing an ID field, including the following:
- OpPersonnelCapabilityType
- OpEquipmentCapabilityType
- OpMaterialCapabilityType

Should ID fields of type IdentifierType be added to these, as well?

Dennis-Brandl, 2020-08-14:
ID fields added in soon to be released version. Issue closed.

closed 2020-08-14 by Dennis-Brandl