MESAInternational/B2MML-BatchML #40: add Required ID to each Op(Resource)SpecificationType and Op(Resource)RequirementType in Common Schema

racheliurui, 2019-09-17:
### Background

Similiar like #39 , a few other types are missing the ID element from definitions.
Common Schema is not aligned with updated 950002, need required ID to each Op(Resource)SpecificationType and Op(Resource)SpecificationType 
```xml
<xsd:element name="ID" 				type="IdentifierType"/>
```

### Supporting Document
Each of the following resource objects has a required ID attribute.
Table 156 – Personnel specification attributes
Table 160 – Equipment specification attributes
Table 164 – Physical asset specification attributes
Table 168 – Material specification attributes
Table 183 – Personnel requirement attributes
Table 187 – Equipment requirement attributes
Table 191 – Physical asset requirement attributes
Table 195 – Material requirement attributes

### Impacted Types

B2MML-Common.xsd

- OpEquipmentRequirementType
- OpPersonnelRequirementType
- OpPhysicalAssetRequirementType
- OpMaterialRequirementType
- OpEquipmentSpecificationType
- OpPersonnelSpecificationType
- OpPhysicalAssetSpecificationType
- OpMaterialSpecificationType

Dennis-Brandl, 2020-01-28:
This is a breaking change, because the ID is required. 
This will be fixed in the January 2020 Sprint.

JoeDo, 2020-02-03:
Confirmed for the following types in the January-2020-Sprint branch:
- OpEquipmentRequirementType
- OpPersonnelRequirementType
- OpPhysicalAssetRequirementType
- OpMaterialRequirementType
- OpPersonnelSpecificationType
- OpEquipmentSpecificationType
- OpMaterialSpecificationType (no change)
- OpPhysicalAssetSpecificationType

I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
To be fixed in the soon to be released version. Issue closed.

closed 2020-08-14 by Dennis-Brandl