MESAInternational/B2MML-BatchML #41: add Required ID to each (Resource)SegmentSpecificationType in ProcessSegment Schema

racheliurui, 2019-09-17:
### Background

Similiar like #39 , a few other types are missing the ID element from definitions.

ProcessSegment Schema is not aligned with updated 950002, need required ID to each (Resource)SegmentSpecificationType.

```xml 
<xsd:element name="ID" 				type="IdentifierType"/>
```

### Supporting Documents

Each of the following resource objects has a required ID attribute.
>Table 80 – Personnel segment specification attributes
Table 84 – Equipment segment specification attributes
Table 88 – Physical asset segment specification attributes
Table 92 – Material segment specification attributes

### Impacted Types
B2MML-ProcessSegment.xsd 

- EquipmentSegmentSpecificationType
- PersonnelSegmentSpecificationType
- PhysicalAssetSegmentSpecificationType
- MaterialSegmentSpecificationType

Dennis-Brandl, 2020-01-28:
This is a breaking change, because the ID is required.
This will be fixed in the January 2020 Sprint.

JoeDo, 2020-02-03:
Confirmed for the following types in the January-2020-Sprint branch:
- MaterialSegmentSpecificationType (no change)

@Dennis-Brandl, there appear to be similarly named types that are missing an ID field, including the following:
- EquipmentSegmentSpecificationType
- PersonnelSegmentSpecificationType
- PhysicalAssetSegmentSpecificationType

Should ID fields of type IdentifierType be added to these, as well?

Dennis-Brandl, 2020-08-14:
IDs added in the soon to be released version. Issue closed.

closed 2020-08-14 by Dennis-Brandl