MESAInternational/B2MML-BatchML #42: add HierarchyScope to OpSegmentDataType to align with ISA-950002

racheliurui, 2019-09-17:
### Background
ADD: element name = "HierarchyScope" to OpSegmentDataType 
```xml
<xsd:complexType name = "OpSegmentDataType">
...
 <xsd:element name = "Description" 		type="DescriptionType" 
							minOccurs="0" maxOccurs="unbounded"/>
<xsd:element name = "HierarchyScope"             type = "HierarchyScopeType"    
                                                        minOccurs = "0"/>
...
</xsd:complexType>
```
### Supporting Document 
>ISA-950002
Table 207 – Segment data attributes

### Impacted Types
B2MML-OperationsPerformanceTypes.xsd

- OpSegmentDataType

Dennis-Brandl, 2020-01-28:
To be fixed in the January 2020 Sprint.

JoeDo, 2020-02-03:
Confirmed that the change has been made in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
To be included in the soon to be released version. Issue closed.

closed 2020-08-14 by Dennis-Brandl