MESAInternational/B2MML-BatchML #43: change element MaterialSublotProperty to MaterialLotProperty for MaterialSubLotType

racheliurui, 2019-09-17:
### Background 
For MaterialSubLotType,
Change MaterialSublotProperty to MaterialLotProperty.

```xml
 <xsd:element name = "MaterialSublotProperty"      type = "MaterialLotPropertyType"	     minOccurs = "0" maxOccurs = "unbounded"/>
```
Change to 
```xml
 <xsd:element name = "MaterialLotProperty"      type = "MaterialLotPropertyType"	     minOccurs = "0" maxOccurs = "unbounded"/>
```

Both material lot and material sub lot are referring the same material lot property model. Thus in ISA-950002,  there's no such model called MaterialSublotProperty, so here the element name changed from MaterialSublotProperty to MaterialLotProperty to reflect the property type.

### Supporting Document
>ISA-950002
>Table 72 – Material sublot relationship roles
>5.7.1 Material model
Figure 12 - Material model 

### Impacted Types
B2MML-Material.xsd

- MaterialSubLotType

Dennis-Brandl, 2020-01-28:
No, the sublot property already uses the property type, but it is a sublot property, not a lot property. 
See in section 5.7.8 "Material sublots may have sublot specific properties. " and 
"EXAMPLE 1	Material sublot properties can be RFID tag IDs or other identification properties, such that each material sublot has a different property value."

Dennis-Brandl, 2020-08-14:
The ISA 95 Part 2 text is inconsistent. The text in 5.7.8 says "Material sublots may have **sublot specific properties**.", but Table 72 used the name "**Material lot property**" as the role.  However Table 72 also has the definition of "The **material sublot properties** map to a corresponding material lot property.":
 
Changed to use the element name MaterialLotProperty in the MaterialSubLotType. 
Issue closed.

closed 2020-08-14 by Dennis-Brandl