MESAInternational/B2MML-BatchML #44: remove TestResult from (Resource)PropertyType

racheliurui, 2019-09-17:
### Background
ISA-950002 changed Test Results to a 1st order object as part of the Operations Test Model. Test Result is a 1st order object  is able to be exchanged/queried and has a relationship to resource actual which is represented as a root attribute

But current B2MML 7.0 still has the TestResult type as an attribute in each instance resource property. The old resource model had a composite relationship for resource properties which did not allow Test Result attribute to be exchanged and queried as a 1st class object.

### Supporting Document
ISA-950002
> Section 5.9.4
Table 104 Test specification attributes
Table 45 – Equipment property relationships
Table 46 – Equipment property attributes
Table 70 – Material lot property relationship roles
Table 71 – Material lot property attributes
Table 55 – Physical asset property relationship roles
Table 56 – Physical asset property attributes
Table 36 – Person property relationship roles
Table 37 – Person property attributes

### Impacted Types
B2MML - Equipment.xsd
- EquipmentPropertyType
B2MML - Material.xsd
- MaterialLotPropertyType
B2MML - PhysicalAsset.xsd
- "PhysicalAssetPropertyType"
B2MML - Personnel.xsd
- "PersonPropertyType",

Dennis-Brandl, 2020-01-28:
I have no idea what is being proposed here.

gerardmoloneyetpartners, 2020-01-28:
Hi Dennis, We will update the issue description to make it clear.

Dennis-Brandl, 2020-08-14:
To be included in the soon to be released version. Issue closed.

closed 2020-08-14 by Dennis-Brandl