MESAInternational/B2MML-BatchML #47: Update TestSpecificationType elements to align with ISA-950002

vitalJP, 2019-10-10:
### Background
Multiple changes to TestSpecificationType to align with ISA-950002

>complexType name = "TestSpecificationType", CHANGE PhysicalSampleRequired to PhysicalSample to align with updated Part 2 OperationsTest Model's Test Specification object.

>complexType name = "TestSpecificationType" element name = "EvaluatedProperty", CHANGE type = "TestSpecificationEvaluatedPropertyType" to "EvaluatedPropertyType" to align with updated Part 2 OperationsTest Model's Test Specification object's relationship attributes.

>complexType name = "TestSpecificationType" , ADD element name = "OperationsTestRequirementID" to align with updated Part 2 OperationsTest Model's Test Specification object's relationship attributes.

>complexType name = "TestSpecificationType" , ADD element name = "TestableObjectID" to align with updated Part 2 OperationsTest Model's Test Specification object's relationship attributes.

### Supporting Document
ISA-950002
> Section 5.9.4
Table 104 Test specification attributes

### Impacted Types
B2MML-TestSpecification.xsd

- TestSpecificationType

Dennis-Brandl, 2020-01-28:
1) Name changed to "PhysicalSample" from "PhysicalSampleRequired".
2) Did not change "TestSpecificationEvaluatedPropertyType" to "EvaluatedPropertyType". Because all of the B2MML schemas operate in the same namespce, the prefix "TestSpecification" was added to all types that are unique to the test specification schema. Previously this convention was not used, and it came back to bite us when we wanted to make changes and had name overlaps. 
3) OperationsTestRequirementID Added
4) TestableObjectID Added
This to be added to the January 2020 Sprint.

JoeDo, 2020-02-03:
Confirmed that (1), (3), and (4) have been made in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
Changes made in B2MML-OperationsText.xsd, 
Did not make change (2) because of reason above.
Made changes (1), (3), (4).
Issue closed.

closed 2020-08-14 by Dennis-Brandl