MESAInternational/B2MML-BatchML #48: Update WorkflowSpecificationInformationType to align with ISA-950004

vitalJP, 2019-10-10:
### Background 
> complexType name = "WorkflowSpecificationInformationType">

CHANGE:  
- element name = "WorkflowSpecificationType"   type = "WorkflowSpecificationTypeType"  TO
     "WorkflowSpecificationNodeType"   type = "WorkflowSpecificationNodeTypeType" 

ADD:
- element name = "WorkflowSpecificationConnectionType"   type = "WorkflowSpecificationConnectionTypeType" minOccurs = "0" maxOccurs = "unbounded"/>

Dennis-Brandl, 2020-01-28:
The proposed change does not match ISA 95 Part 4. The Node is just a step in a workflow, the WorkflowSpecificationInformation contains a set of WorkflowSpecifications, which contain nodes and connections. 
The WorkflowSpecificaionType is missing the relationship to other IncludesThePropertiesOf
Not shown in ISA 95 Part 4 is a WorkflowSpecificationType, which NodeSpecifications and ConnectionSpecifications.  This was added as the WorkflowSpecificationTypeType.

JoeDo, 2020-02-03:
Confirmed the addition of an IncludesThePropertiesOf element in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
Changed in the soon to be released version, Issue closed.

closed 2020-08-14 by Dennis-Brandl