MESAInternational/B2MML-BatchML #50: Add XSD ComplexType name "SpatialDefinitionFormatType" to Common Schema as enumerations to align to 950002, Clause 5.2

vitalJP, 2019-10-10:
Added to Common Schema. 
XSD ComplexType name "SpatialFormatDataType" as enumerations to comply and align to ISA-95 Part 2 specification, section 5.2 Spatial definitions on Table 19 with enumerator format data types. 
There are 8 possible common format types with spatial data identification type acronyms. These include WKT, WKB, GML, KML, GPX, GeoJSON, SVG, and other. The documentation of these types are listed above in the XSD schema complex type.

The Spatial Format Enumeration Data Types are as follows:
1. WKT - Well Known Text ! markup language defined in the ISO/IEC 13249-3:2011 Standard
2. WKB - Well Known Binary ! representation of WKT defined in the ISO/IEC 13249-3:2011 Standard
3. GML - Geography markup language - OpenGIS Geography Markup Language Encoding Standard
4. KML - Keyhole markup language - KML 2.2 defined by the Open Geospatial Consortium (OGC)
5. GPX - GPS exchange format data used from mobile devices.
6. GeoJSON - Geographical format for spatial data in JSON format.
7. SVG - Scalable Vector Graphics in XML with 2D and 3D lines, poly lines, or solid objects.

```xml
   <xsd:complexType name="SpatialDefnitionFormat1Type">

	<xsd:simpleContent>

	    <xsd:restriction base="CodeType">

		<xsd:enumeration value="WKT"/> <!--Well-Known Text ISO/IEC 13249-3:2011 Standard -->  

	        <xsd:enumeration value="WKB"/> <!--Well-Known Binary ISO/IEC 13249-3:2011 Standard --> 

                <xsd:enumeration value="GML"/> <!--Geographical Markup Language  --> 

                <xsd:enumeration value="KML"/> <!--Keyhole Markup Language  --> 

                <xsd:enumeration value="GPX"/> <!--GPS Exchange Format  --> 

                <xsd:enumeration value="GeoJSON"/> <!--Geographical JSON 1.0 Format  --> 

                <xsd:enumeration value="SVG"/> <!--Scalable Vector Graphics XML 2D or 3D format -->

		<xsd:enumeration value="Other"/> <!--Other undefined Spatial Unknown format --> 

            </xsd:restriction>

        </xsd:simpleContent>

    </xsd:complexType>	

    <!--     --> 

    <xsd:complexType name="SpatialDefnitionFormatType">

		<xsd:annotation> 

			<xsd:documentation>

For the SpatialDefintionType, the Spatial Defintion Format element's Enumeration types as defined in ISAS-95 Part 2 commom object models Spatial data definitions.    -->

			</xsd:documentation>

		</xsd:annotation> 

         <xsd:simpleContent>

            <xsd:extension base="SpatialDefnitionFormat1Type">

                <xsd:attribute name="OtherValue" type="xsd:string"/>

            </xsd:extension>

        </xsd:simpleContent>

    </xsd:complexType>	
```

Dennis-Brandl, 2020-01-28:
These are only example in ISA 95 Part 2. They should not be hard coded in B2MML.
Recommend reject.

Dennis-Brandl, 2020-08-14:
Those are still just examples, but its ok to leave this change in place, and let people use the "OtherValue" where needed.  
Issue closed.

closed 2020-08-14 by Dennis-Brandl

gerardmoloneyetpartners, 2020-09-07:
I see this is committed now to the master. Thanks.
1 comment though, the code could use some clean up. there are unnecessary empty comments left throughout the entire common.xsd     __

[image]

Dennis-Brandl, 2020-09-07:
Yes, there was a mixture of tabs and spaces, so the indentation was messed up on the one I pulled grit ETPartners. Some of the editing was previulsy done with tabs set to 2 spaces, some with tabs to 4 spaces, and some with just spaces.  I fixed a lot of them on the MESA version, but did not get all of them.  

 

I left the  because they serve to help make the schema more readable using something like Notepad++. It doesn’t hurt to have them there. 

Dennis

 

Dennis Brandl

208 Townsend Ct, Suite 200

Cary, NC 27518-8319, USA

+1-919-852-5322 (Office)

+1-919-656-2205 (Cell)

+1-832-201-0554 (Fax)

 <http://www.brlconsulting.com/> www.brlconsulting.com   

    <mailto:DnBrandl@BRLConsulting.com> DnBrandl@BRLConsulting.com 

 

 <https://user-images.githubusercontent.com/54049079/92350066-6d10b580-f10a-11ea-9999-afaa02e0f3dd.PNG>