MESAInternational/B2MML-BatchML #51: Update all ID element attributes in B2MML Schema and Types as "required" to align with updated 950002 and 950004.

vitalJP, 2019-10-10:
**Background**
The updated 950002 and 950004 has defined all ID attributes on objects as "required" or mandatory.  Many B2MML schema and types have the IDs as optional (0..1)
element name = "ID" type = "IdentifierType"  minOccurs = "0"/>.
This needs to be changed to:
   TO:  element name = "ID" type = "IdentifierType" />

**Supporting Document**
Multiple resource objects must have a required ID attribute.
All object attribute tables in 950002 and 950004.

PierreREN, 2019-10-13:
I found the usage of "IdentifierType" very confusing as well. 

When checking on the detailed description  in standard document 950002.4.5.2 or 950004.5.4.2 "Object identification" part, relevant instructions are given:
1. These IDs shall be unique within the scope of the exchanged information, requiring translations from the internal ID of the source system to the interface content ID and from the interface content ID to the internal ID of the target system;
2. A unique identification set shall be agreed to in order to exchange information. 
3. The object IDs are used only to identify object within related exchanged information sets. The object ID attributes are not global object IDs or **_database index attributes_**.
4. Generally, objects that are elements of aggregations, and are not referenced elsewhere in the model, do not require unique IDs.

Thus, from my point of view, the IdentifierType is equivalent to the "unique IDs" mentionned above, which means it is not always necessary when the 4th term describes the case. But as the IdentifierType is used everywhere a refenrence of other objects exist, was IdentifierType designed to be the general primary key for every object? And if so, how should we use IdentifierType as general primary key when it is composed of attributes like value, schemaID, schemaName, schemaAgencyID... Or maybe only "value" attribute is needed as primary key and IdentifierType is regarded as an "Embeddable" type to faciliate persistence. Only when tranferring data from one source to another, the instance of original IdentifierType of B2MML is generated by implementing the atrributes with "value" and making up other attributes.

Can @Dennis-Brandl kindly explicate the correct use method of this class from the designer's view? Thanks so much.

Dennis-Brandl, 2020-01-28:
This needs to go to the ISA 95 committee. 
B2MML reflects the standard. 
Recommend closing this issue.

gerardmoloneyetpartners, 2020-02-03:
Hi Dennis, since publishing this issue we may already have reviewed with ISA95 committee.
Will post an update with the feedback from ISA95 once we review again.

Dennis-Brandl, 2020-08-14:
Issue closed. No updates since February.  No actions. 
IDs are not intended to be "primary keys".  They are only valid for the duration of the exchanges (which may include multiple transactions).  See ISA 95 Part 7 and ISA TR95.01 for further definitions.
Issue closed.

closed 2020-08-14 by Dennis-Brandl

gerardmoloneyetpartners, 2020-09-07:
Hi @Dennis-Brandl 
I did not get back to you before you closed this issue.
since then, I did provide you with the references to the October 2019 ISA95 meeting references where there was discussion on certain changes. I think you should review this change in light of that information to see if there was direction from the discussion with ISA95 regarding this proposed change.