MESAInternational/B2MML-BatchML #52: Add ActionType and Action1Type to Common.xsd for defined values of Action element in "OperationsRecordSpecTemplateType" and "OperationsRecordTemplateType"

vitalJP, 2019-10-10:
To align with Updated 95002, Operations Record Model
**Solution**
1.  complexType name="OperationsRecordSpecTemplateType"   element name="Action"
   CHANGE:  type="IdentifierType" 	minOccurs="0" maxOccurs="unbounded"/>
    TO:  type="ActionType"  minOccurs="0"/>

2.  complexType name="OperationsRecordType"   element name="Action"
   CHANGE:  type="IdentifierType" 	minOccurs="0" maxOccurs="unbounded"/>
    TO:  type="ActionType"  minOccurs="0"/

3.  ADD ActionType and Action1Type to Common.xsd for defined values of Action element in "OperationsRecordSpecTemplateType" and "OperationsRecordSpecTemplateType" 
```xml
  <xsd:complexType name="Action1Type">

        <xsd:simpleContent>

            <xsd:restriction base="CodeType">

                <xsd:enumeration value="Added"/>

                <xsd:enumeration value="Deleted"/>

                <xsd:enumeration value="Changed"/>

                <xsd:enumeration value="Observed"/>

                <xsd:enumeration value="Other"/>

            </xsd:restriction>

        </xsd:simpleContent>

    </xsd:complexType>

    

    <!--     -->    

    <xsd:complexType name="ActionType">

		<xsd:annotation> 

			<xsd:documentation>

The permitted set of actions applied to the operations record entry template object in the operations record template by the publisher.

Defined values for action are

added, changed, deleted, and observed.

If no action is specified, this is equivalent to all actions being allowed.

			</xsd:documentation>

		</xsd:annotation> 

         <xsd:simpleContent>

            <xsd:extension base="Action1Type">

                <xsd:attribute name="OtherValue" type="xsd:string"/>

            </xsd:extension>

        </xsd:simpleContent>

    </xsd:complexType>

	
```

**Supporting Document**
ISA-950002
Clause 5.10.2 Operations record specification template (abstract), Table 119 – Operations record specification template attributes
Clause 5.10.3 Operations record template, Table 121 – Operations record template attributes

**Impacted types**
B2MML-Common.xsd: OperationsRecordSpecTemplateType
B2MML-Common.xsd: OperationsRecordTemplateType

Dennis-Brandl, 2020-01-28:
Implemented as OperationsRecordActionType, instead of just ActionType. We have found that adding adjectives in front of common words is really helpful.  
This will be part of the January 2020 Sprint.

JoeDo, 2020-02-03:
Confirmed that the changes have been made in the January-2020-Sprint branch.

@Dennis-Brandl, contrary to the original proposal, the maxOccurs attributes are still present in the type definitions, but I assume this is intentional to be consistent with the other elements.

I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
Implemented as OperationsRecordActionType, because of above reason. 
Issue closed.

closed 2020-08-14 by Dennis-Brandl