MESAInternational/B2MML-BatchML #53: Add "OperationsRecordSpecTemplateID" and "OperationsRecordEntryTemplateID" elements to "OperationsRecordTemplateType" in Common.xsd to align with updated 950002 Operations Record Model

vitalJP, 2019-10-10:
Add "OperationsRecordSpecTemplateID" and "OperationsRecordEntryTemplateID" elements to  "OperationsRecordTemplateType" in Common.xsd to align with updated 950002 Operations Record Model

**Solution** 
"OperationsRecordTemplateType"  
ADD element name="OperationsRecordEntryID"	type="IdentifierType" minOccurs="1" maxOccurs="unbounded"/>
ADD element name="OperationsRecordSpecTemplateID"

**Support Documents**
950002
Clause 5.10.3 Operations record template, Table 120 – Operations record template relationship roles

**Impacted Types**
B2MML- Common.xsd,  complexType name="OperationsRecordTemplateType"

Dennis-Brandl, 2020-01-28:
The relationship is a composite, so instead of adding an ID, I added the following element under the OperationsRecordTemplateType:
<xsd:element name="RecordEntry' type="OperationsRecordEntryTemplateType"
  minOccurs="0" maxOccurs="unbounded"/>

Also found that the OperationsEvent/OperationsEventRecord was the wrong type. It was OperationsRecordEntryTemplateType, and should have been OperationsRecordTemplateType.

JoeDo, 2020-02-03:
Confirmed that both changes have been made in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
Changed as marked above.
Issue closed.

closed 2020-08-14 by Dennis-Brandl