MESAInternational/B2MML-BatchML #54: Add ExtensionType and Extension1Type to Common.xsd for defined values of Extension element in "OperationsEventClassType"

vitalJP, 2019-10-10:
To align with Updated 95002, Operations Event Model
**Solution**
complexType name="OperationsEventClassType" element name="Extension"
CHANGE: type="IdentifierType" minOccurs="0" maxOccurs="unbounded"/>
TO: type="ExtensionType" minOccurs="0"/>

ADD ExtensionType and Extension1Type to Common.xsd for defined values of Extension element in "OperationsEventClassType" to align with Updated 950002, Operations Event Model
```xml
 <xsd:complexType name="Extension1Type">

        <xsd:simpleContent>

            <xsd:restriction base="CodeType">

                <xsd:enumeration value="Sealed"/>

                <xsd:enumeration value="NA"/>

                <xsd:enumeration value="Other"/>

            </xsd:restriction>

        </xsd:simpleContent>

    </xsd:complexType>

    

    <!--     -->    

    <xsd:complexType name="ExtensionType">

		<xsd:annotation> 

			<xsd:documentation>

Defines constraints on the inclusion of the respective operations event class in a hierarchy. 

Defined value are 

-sealed: no entry can use this entry as its parent;

-(No entry): no entry allows any entry to represent this entry as its parent /ancestor.

Extension can contain a set of entries of other operations event classes who represent this entry in their parent.

			</xsd:documentation>

		</xsd:annotation> 

         <xsd:simpleContent>

            <xsd:extension base="Extension1Type">

                <xsd:attribute name="OtherValue" type="xsd:string"/>

            </xsd:extension>

        </xsd:simpleContent>

    </xsd:complexType>	
```

**Supporting Document**
ISA-950002
Clause 5.11.2 Operations event class, Table 126 – Operations event class attributes

**Impacted types**
B2MML-OperatrionsEvent.xsd: OperationsEventClasssType
New B2MML-Common.xsd: ExtensionType
New B2MML-Common.xsd: Extension1Type

Dennis-Brandl, 2020-01-28:
Renamed to codetype to OperationsEventExtensionType and put it into the OperationsEvent.xsd file. 
To be fixed in January 2020 Sprint.

JoeDo, 2020-02-03:
Confirmed the addition of the OperationsEventExtensionType in OperationsEvent.xsd in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
Changed as specified above.
Issue closed.

gerardmoloneyetpartners, 2020-08-15:
how do I see your commit for this change? its not listed here?

Dennis-Brandl, 2020-08-17:
Changed name to "OperationsEventExtension1Type" as specified in Jan 28th comment.

Dennis-Brandl, 2023-02-05:
I started rebuilding the next version, and I am not sure what happended to this change. Let's leave it open until we find the right changed source.