MESAInternational/B2MML-BatchML #55: Delete defined values in ClassPropertyTypeType in Common.xsd that are not defined in Updated 950002

vitalJP, 2019-10-10:
To align with defined values for property type attributes in Updated 950002  
**Solution**
Common.xsd, complexType name="ClassPropertyTypeType"
DELETE:    <xsd:enumeration value="Inventory"/>
                <xsd:enumeration value="Mixed"/
These two enumerations are not supported by the updated 950002.

**Supporting Document**
All references where property type attribute is used in ISA-950002
5.5.3	Equipment class property, Table 42 – Equipment class property attributes
5.4.3	Personnel class property, Table 33 – Personnel class property attributes
5.6.3	Physical asset class property, Table 52 – Physical asset class property attributes
5.7.3	Material class property, Table 63 – Material class property attributes
5.7.5	Material definition property, Table 67 – Material definition property attributes
5.3.3	Operational location class property, Table 24, – Location class property attributes
5.11.3 Operations event class property, Table 129 – Operations event class property attributes
5.11.6 Operations event definition property, Table 134 – Operations event definition property attributes

**Impacted types**
B2MML-Common.xsd: ClassPropertyTypeType

Dennis-Brandl, 2020-01-28:
Typo,
To be fixed in the January 2020 Sprint.

JoeDo, 2020-02-03:
Confirmed that the change has been made in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
Changed as above. 
Issue closed.

closed 2020-08-14 by Dennis-Brandl