MESAInternational/B2MML-BatchML #56: Add OperationsEventClassID to OperationsEventDefinitionType in OperationsEvent.xsd to align with updated 950002

vitalJP, 2019-10-10:
complexType name = "OperationsEventDefinitionType"> to align with update 950002, Operations Event Model

**Solution**
ADD element name = "OperationsEventClassID    type = "IdentifierType"  minOccurs = "0" maxOccurs = "unbounded"/>

**Supporting Document**
950002, Clause 5.11.5 Operations event definition, Table 131 – Operations event definition relationship roles

**Impacted Types**
B2MML OperationsEvent.xsd:  "OperationsEventDefinitionType"

Dennis-Brandl, 2020-01-28:
To be fixed in January 2020 Sprint

JoeDo, 2020-02-03:
Confirmed that the change has been made in the January-2020-Sprint branch. I'll circle back and Close this issue after the changes have been merged into Master.

Dennis-Brandl, 2020-08-14:
Fixed as above in the soon to be released version. 
Issue closed.

closed 2020-08-14 by Dennis-Brandl