MESAInternational/B2MML-BatchML #57: Add Generic "B2mmlInformation" Element

bebedizo, 2019-10-14:
Maybe this feature exists, and I haven't found it (please, point me to the right source if it exists):
## Issue
It would be nice to be able to model multiple kinds of `*Information` in a single B2MML file. Currently, the Schemas are designed for modeling, e.g., `EquipmentInformation` in a separate file, and `MaterialInformation` in another file.
## Solution
Why not create (e.g., in `B2MML-Common.xsd`) a "parent" element named `B2mmlInformation` that contains 0..1 entries of kind `PersonnelInformation`, then 0..1 entries of kind `EquipmentInformation`, etc.

Is there a specific argument against such a "super file"?

Dennis-Brandl, 2020-01-20:
We are currently implementing a new schema based on the ISA 95 TR.01 - Profile Template, which provides a single file that contains all of the information that could be used setup (configure) a system.  Dave and I hope to have this posted this month.  It basically includes all of the equipment, material, personnel, information in a single file, and also includes such things as units of measure, data types, enumerations, etc ...

gerardmoloneyetpartners, 2020-01-21:
Hi Bebedizo,
The current files on MESA are the baseline files.
You can design your own .xsd as a wrapper to include any or all of the baseline schema.
This is viewed in ISA95, for each particular customer implementation, as an OperationsDefinedEvent that creates templates that the integrated systems need to use to communicate back and forth.
You can design those templates to combine the files as you are suggesting.

It's been done before on customer specific implementation projects, but is not part of the MESA files until the team complete a similar task (I assume as a generic example) as Dennis has mentioned.

gerardmoloneyetpartners, 2020-01-21:
^^ take a look at OperationsRecordInformation and OperationsEventDefinitionRecordSpecification in ISA95 Part 2

Dennis-Brandl, 2020-01-28:
Please check out the B2MML-MasterDataProfile.xsd file and the associated ISA TR95.01 Master Data Profile Template. 
This can be used for the master data. it does not cover material lots or sublots or persons.  But it does cover Equipment and Equipment class.

closed 2020-01-28 by Dennis-Brandl