MESAInternational/B2MML-BatchML #58: Unable to generate java classes with valid enumeration facets via xjc

PierreREN, 2019-12-08:
When generating java classes via xjc command of jaxb, the enumetation facets cannot be correctly inherite in the generated java classes. For example, the "EquipmentElementLevel1Type" extends "CodeType" and is designed to be an enumeration of all the valid values when describing the equipment element hierarchy level, as below: 

<xsd:complexType name="EquipmentElementLevel1Type">
        <xsd:simpleContent>
            <xsd:restriction base="CodeType">
                <xsd:enumeration value="Enterprise"/>
                <xsd:enumeration value="Site"/>
                <xsd:enumeration value="Area"/>
                <xsd:enumeration value="ProcessCell"/>
                <xsd:enumeration value="Unit"/>
                <xsd:enumeration value="ProductionLine"/>
                <xsd:enumeration value="WorkCell"/>
                <xsd:enumeration value="ProductionUnit"/>
                <xsd:enumeration value="StorageZone"/>
                <xsd:enumeration value="StorageUnit"/>
                <xsd:enumeration value="WorkCenter"/>
                <xsd:enumeration value="WorkUnit"/>
                <xsd:enumeration value="EquipmentModule"/>
                <xsd:enumeration value="ControlModule"/>
                <xsd:enumeration value="Other"/>
            </xsd:restriction>
        </xsd:simpleContent>
    </xsd:complexType>

However, the generated java class indicates no enumeration facets:

@XmlAccessorType(XmlAccessType.FIELD)
@XmlType(name = "EquipmentElementLevel1Type")
@XmlSeeAlso({
    EquipmentElementLevelType.class
})
public class EquipmentElementLevel1Type
    extends CodeType
{

}

How can the enumeration facets be executed in this case?

PierreREN, 2019-12-28:
@JoeDo @racheliurui @BrentHumphreys @Emers @racheliurui 
Thanks so much for your explanation on the usage of CodeType as a unique tool for dealing with enumerations. This point has been confusing me a lot. Haven't understood the inspiration why you guys design such a type.

Dennis-Brandl, 2020-01-28:
The type was used so that the "Other" type could have the additional element "OtherValue" to contain the string of the other value when the "Other" enumeration is used. This was the recommended method when these were built. If there is a better way (which is backward compatible), then we could use it.  Any suggestions are welcome.

PierreREN, 2020-02-24:
I have been seeking for alternative ways for the enumerative types, and it seems that I always prefer the simplest way, which is to use the enumeration classes directly without extending the CodeType class. The enumeration classes will explicate themselves when transferred, as the enumeration classes can be interpreted when defined as a simple type based on String.
However, in my opinion, the most important idea of CodeType is to indicate not only the enumeration list but also other related information when transferring the data to another system. And other classess such as IdentifierType are of the same case. From this point, the current design seems quite reasonable and better ways (which is especially backward compatible) are not that obvious. 
I may let go this issue for now and keep thinking about other possibilities in the future. However, to face my current concern, I tend to give up the usage of classes like IdentifierType or CodeType and drop part of the B2MML termilogies from my major project. The B2MML types will more likely to be used as the interfaces among systems, dealing with DTOs, rather than to be relied on as a basis for designing new MOM-related applications, dealing with domain entities. Perhaps that is more close to what you expect B2MML to be when first designed, is it?

closed 2020-02-24 by PierreREN