MESAInternational/B2MML-BatchML #59: Please Review Branch of January 2020 Sprint

Dennis-Brandl, 2020-01-28:
The January 2020 Sprint branch has the following issues addressed. 
Please review. 
We would like to get this version out, for good, and make it the official version for the latest ISA 95 version. 
Addressed Issues:
15, 16, 17, 18, 19, 20, 21, 23, 24, 25, 26, 30, 31, 32, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 47, 48, 50, 51, 52, 53, 54, 55, 56, 58

JoeDo, 2020-02-03:
@Dennis-Brandl, I have gone through all open issues and tagged you in a handful that require your additional input. Once those changes have been incorporated, I will merge your changes into the main branch.

Dennis-Brandl, 2020-08-13:
Ok, I have taken the latest version, checked it against XMLSply, cleaned up the remaining issue I had on the documentation and I am pushing these to the MESA GitHub master.

closed 2020-08-13 by Dennis-Brandl