MESAInternational/B2MML-BatchML #66: Work directive attributes missing

ArnoClaeys, 2020-09-16:
The _workflow specification_ and _job order id_ attributes are missing in the _work directive_ object.

Dennis-Brandl, 2020-09-28:
There is a _JobOrderSourceID_ that contains the ID of of the _job order_.  
I think I remember discussions that the _workflow specification_ is not needed, because the _WorkDirective_ contains the ID of the Work Master, which contains the Workflow Specification.  Can anyone remember if this was the reason that the _WorkflowSpecification_ was not included in the _WorkDirective_?

erwindl0, 2020-09-28:
I would prefer to have a dedicated association between WorkDirective and WorkflowSpecification, to support the situation where a concrete WorkDirective can contain slight deviations in its WorkflowSpecification, compared to the "template" defined via the WorkMaster.

Dennis-Brandl, 2020-09-28:
That is a good point. In that case it would be better to move it to the _WorkDefintion_ because it is common to the _WorkMaster_ and _WorkDirective_.

erwindl0, 2020-09-28:
Indeed. 

Something similar seems the case for JobOrderSourceID?

But I noticed that in WorkMaster it is not optional, while in WorkDirective it is optional (0..1).
Personally I would assume that WorkMasters can exist without JobOrders (so JobOrderSourceID should be optional there as well).

A concrete JobOrder may or may not have a WorkDefinition. 
If it has one, it could be a WorkMaster (if the template is sufficient) or a derived WorkDirective.

So maybe a choice element on JobOrderType to point to either a pair of WorkMasterID/WorkMasterVersion or a WorkDirectiveID/WorkDirectiveVersion could be an option? (and WorkDirective should have not only the WorkMasterID but also the WorkMasterVersion from which it was derived?)

And WorkDefinition could contain an optional JobOrderSourceID?

(disclaimer: I just started learning the details of ISA-95 and B2MML since a year or so, so I may have missed lots of previous discussions/decisions on these topics)

Dennis-Brandl, 2020-09-30:
Ok, I took a longer look at this and checked against ISA-95 Part 4.  
1) Yes, the _WorkMasterType_ / _JobOrderSourceID_ should be minOccurs="0",
2) Yes, the _WorkDirectiveType_ should have a _WorkflowSpecification_ element with minOccurs="0". 
3) Both of these could go into the _WorkDefinition_ because they have the same rules.  However, the ISA-95 model shows these as separate relationships under the _WorkMaster_ and _WorkDirective_.  From the B2MML XML viewpoint, it does not matter where they are defined.  So in order to keep things simple I recommend we move the _JobOrderSourceID_ and the _WorkflowSpecification_ into the WorkDefinition.  This is a breaking change (but there are so many in Version 7, this is minor compared to other changes.) 
Thoughts?

ArnoClaeys, 2020-09-30:
I think this is a good solution. If I’m correct, the _WorkflowSpecification_ was located in the _WorkDefinition_ in Version 6. 

As stated by Erwin above, a work directive starts as a copy of a work master. Still, some adjustments in the workflow specification can occur based on the job order requirements (for example product variant or equipment used). Also, in this scenario workflow specification nodes may refer to another work directive. 

Many thanks for the effort!

Dennis-Brandl, 2020-10-02:
Unless I hear any objections from anyone over the next few days (until Oct 7th). I will commit these changes to the master on the 8th. 
@ -143,6 +143,10 @@
                                                        minOccurs = "0" maxOccurs = "unbounded"/>
      <xsd:element name = "MaterialSpecification"       type = "OpMaterialSpecificationType"
                                                        minOccurs = "0" maxOccurs = "unbounded"/>
      **_<xsd:element name = "JobOrderSourceID"            type = "IdentifierType"
                                                        minOccurs = "0"/>
      <xsd:element name = "WorkflowSpecification"       type = "WorkflowSpecificationType"
                                                        minOccurs = "0"/>_**
    </xsd:sequence>
  </xsd:group>

@ -150,11 +154,8 @@
  <xsd:complexType name = "WorkMasterType">
    <xsd:sequence>
      <xsd:group   ref  = "WorkDefinitionType"/>
      _<xsd:element name = "JobOrderSourceID"            type = "IdentifierType"/>_
      <xsd:element name = "WorkMasterChild"             type = "WorkMasterType"
                                                        minOccurs = "0" maxOccurs = "unbounded"/>
      _<xsd:element name = "WorkflowSpecification"       type = "WorkflowSpecificationType"
                                                        minOccurs = "0"/>_
      <xsd:element name = "OperationsDefinitionID"      type = "IdentifierType"
                                                        minOccurs = "0" maxOccurs = "unbounded"/>
      <xsd:element name = "OperationsSegmentID"         type = "IdentifierType"
@ -173,8 +174,6 @@
    <xsd:sequence>
      <xsd:group   ref  = "WorkDefinitionType"/>
      <xsd:element name = "WorkMasterID"                type = "IdentifierType"/>
      _<xsd:element name = "JobOrderSourceID"            type = "IdentifierType"
                                                        minOccurs = "0"/>_
      <xsd:element name = "WorkDirectiveChild"          type = "WorkDirectiveType"
                                                        minOccurs = "0" maxOccurs = "unbounded"/>
      <xsd:element name = "WorkDirectiveState"          type = "ResponseStateType"

closed 2020-10-28 by Dennis-Brandl