MESAInternational/B2MML-BatchML #74: Change OpEquipmentRequirementType\EquipmentLevel codetype to EquipmentLevelType

gerardmoloneyetpartners, 2021-06-03:
The B2MML-Common.xsd defines the codetype for  OpEquipmentRequirementType\EquipmentLevel codetype=HierarchyScopeType

This should be CodeType=EquipmentLevel

Additionally, the Sequence should relocate the EquipmentLevel to be immediately after the EquipmentID for easier understanding.

Existing XSD (lines 893 to 912) with the issue at Line 904
[image]

Proposed Fix - Relocate to line 898, and change codetype=EquipmentLevelType
[image]

h28669, 2021-06-14:
I agree. It looks like wrong type in current schema definition.

ETPartners, 2023-01-27:
Any movement on this please guys?

gerardmoloneyetpartners, 2023-01-27:
Any movement on this guys?

Dennis-Brandl, 2023-02-04:
Yes, they need to be changed in the next version.

Dennis-Brandl, 2023-02-09:
Fixed with latest merge with main

closed 2023-02-09 by Dennis-Brandl