MESAInternational/B2MML-BatchML #75: The enumeration "ResourceLocationType1Type" repeats the "Equipment" field

BigVulcanDeal, 2021-11-29:
In B2MML-Common.xsd, the ResourceLocationType1Type repeats the field "Equipment

    <xsd:complexType name="ResourceLocationType1Type">
        <xsd:simpleContent>
            <xsd:restriction base="CodeType">
                <xsd:enumeration value="Operational Location ID"/>
                <xsd:enumeration value="Operational Location"/>
                <xsd:enumeration value="Equipment"/>
                <xsd:enumeration value="Person"/>
                <xsd:enumeration value="Equipment"/>
                <xsd:enumeration value="Physical Asset"/>
                <xsd:enumeration value="Description"/>
                <xsd:enumeration value="Other"/>
            </xsd:restriction>
        </xsd:simpleContent>
    </xsd:complexType>

BigVulcanDeal, 2021-11-29:
After looking a little closer, I assume that one of these "Equipment" entries should be something else .. "Material" comes to mind ..

gerardmoloneyetpartners, 2023-01-27:
Actually, if we read the documentation a few lines down, it explains the enumeration list. 
[image]

In all likelihood, the list was probably copied from somewhere, and instead of deleting the 'material' its been changed to 'equipment' and introduced the duplicate in the list. 

I vote for deleting line 1572 to remove the duplicate.
There is no need to reference something being stored in, or required to move inside a MaterialLot. So a Material reference in the ResourceLocationType1Type is not required, as per the documentation.

Dennis-Brandl, 2023-02-04:
I will remove.

closed 2023-02-09 by Dennis-Brandl