MESAInternational/B2MML-BatchML #78: B2MML-JSON: some findings/issues

thucnguyen77, 2022-10-13:
Even though I have successfully generated the C# POCO model from the JSON schema. Many issues I had to fix in order to bring it up to a usable level.

- The prefix `.b2mml` or `b2mml` doesn't look helpful so I had to remove.
- Many duplicated types. They have the same structure but slight different names e.g. `DataSetType` and `DataSet1Type`.
- Redundant enum value, e.g. `DateTime` and `dateTime`.
- Typos, such as `MaterialSublot` at line 22171, `MaterialSublotID` at line 23680. This would generate incorrect C# class name.

Hope someone can have a look and fix if the findings are valid.

Dennis-Brandl, 2023-02-04:
Comments:
1) The .b2MML and b2mml were automatically added by the XML Spy tool used.  Updated the version to remove these two prefixes. No problems notes in XMLSpy 2021 or Visual Studio 2022.
2) Duplicated types were because of the combination of B2MML and BatchML and conversion from XML.  Not sure it is worth the effort to fix the JSON, since it will become out of sync with the XML.
3. The DateTime is a core component type that contains attributes, the dateTime is a basic XML type, ithout the associated UN/CEFACT attributes.
4.  I am not sure what the typos are.

Dennis-Brandl, 2023-02-08:
I used XML Spy to convert the XML schema to a JSON Schema, using the the "http://json-schema.org/2020-12/schema"
I removed the b2mml: and .b2mml: prefixes. 
I imported the JSON schema into Visual Studio 2022 using the Edit -Paste Special - Paste JSON as Classes.
There is still a namespace conflict for the definitions of "Othervalue", "Name", "Name1", and Format".  These would have to be manually fixed, unless someone has a simple solution.

Dennis-Brandl, 2023-02-09:
I will close this, and ask for a new issue to be opened on the new version, if the typos still exist.  I couldn't find them. This JSON version also uses the new 2020-12 JSON Schema, so it should be checked for additional problems.

closed 2023-02-09 by Dennis-Brandl