MESAInternational/B2MML-BatchML #80: Add WorkDefinition extensions group

ArnoClaeys, 2023-02-10:
As in some cases, if not in most cases, extensions on WorkMaster also apply on WorkDirective, it would be convenient to also add a extensions group to WorkDefinition.

See proposed commit: https://github.com/MESAInternational/B2MML-BatchML/pull/77/commits/77de6350d61aeac233cf7463a3f8e07919537553

Dennis-Brandl, 2023-02-14:
Good idea, something we missed.
We have an extension in the WorkDefinitionInformation, which is a container object.
We have extensions in WorkMasters and in WorkDirectives.
This changes adds a common extension to the shared information between all WorkMasters and WorkDirectives.
This is not a breaking change. See three types of extensions in figure below.
image
[image]

ArnoClaeys, 2023-03-16:
Looks good. Thanks!

closed 2023-03-16 by ArnoClaeys