MESAInternational/B2MML-BatchML #88: EquipmentLevel for PhysicalAssetType is defined as HierarchyScopeType

alexmarginet-etp, 2023-06-06:
The Type of the `EquipmentLevel` in `PhysicalAssetType` and `OpPhysicalAssetRequirementType` is set to `HierarchyScopeType`, but it should be `EquipmentLevelType`

See reference lines:

https://github.com/MESAInternational/B2MML-BatchML/blob/075a1ed579a0e795b4e82ceeb4a0a0e7fa157fea/Schema/B2MML-PhysicalAsset.xsd#LL170C28-L170C42

https://github.com/MESAInternational/B2MML-BatchML/blob/075a1ed579a0e795b4e82ceeb4a0a0e7fa157fea/Schema/B2MML-Common.xsd#L981

alexmarginet-etp, 2023-06-06:
Sorry, I have just seen that this is already reported and explained in issue #32

Dennis-Brandl, 2023-07-11:
Fixed July 11, 2023

gerardmoloneyetpartners, 2026-01-06:
Have re-opened the issue, as it was missed in the PhysicalAsset.xsd
Line170 still needs the type updated to 'EquipmentLevelType"

[image]